raspishake/rsudp #1: Freezing after approximately 5 min

andrewkochie, 2019-11-25:
Running latest version of linux mint and used the instructions. I connect to R2DE5 over a vpn remotely to access the shake with no problems. Seems after running the script after 5 min from bash it will freeze with no warning.

iannesbitt, 2019-11-26:
@andrewkochie after this happens, can you quit the program, then attach the file at `/tmp/rsudp/rsudp.log`?

andrewkochie, 2019-11-26:
2019-11-26 19:57:20 [Main] Logging initialized successfully.
2019-11-26 19:57:20 Using settings file: /home/kochie/.config/rsudp/rsudp_settings.json
2019-11-26 19:57:20 Output directory is: /var/www/html/R2DE5
2019-11-26 19:57:20 [Init] Initializing.
2019-11-26 19:57:20 [openSOCK] Opening socket on localhost:8888 (HOST:PORT)
2019-11-26 19:57:20 [Init] Waiting for UDP data on port 8888...
2019-11-26 19:57:22 [Init] Available channels: ['EHZ', 'ENZ', 'ENN', 'ENE']
2019-11-26 19:57:22 [get_inventory] No station name given, continuing without inventory.
2019-11-26 19:57:22 [Plot] Starting.
2019-11-26 19:57:22 [Plot] Plotting channels: ['EHZ']
2019-11-26 19:57:22 [Plot] Seismogram units are Voltage counts
2019-11-26 19:57:22 [Alert] Alert stream units are Voltage counts
2019-11-26 19:57:22 [Alert] Starting Alert trigger with sta=6s, lta=30s, and threshold=1.7 on channel=?HZ
2019-11-26 19:57:22 [Consumer] Starting.
2019-11-26 19:57:22 [Producer] Starting.
2019-11-26 19:57:22 [Producer] Receiving UDP data from 10.147.17.88
2019-11-26 19:57:22 [Alert] Listening to channel EHZ
2019-11-26 19:57:22 [Alert] Earthquake trigger warmup time of 30 seconds...
2019-11-26 19:57:52 [Alert] Earthquake trigger up and running normally.
2019-11-26 19:58:02 [Producer] Sending TERM signal to threads...
2019-11-26 19:58:02 [Consumer] Exiting.
2019-11-26 19:58:02 [Alert] Exiting.
2019-11-26 19:58:03 [Plot] Exiting.
2019-11-26 19:58:03 [Main] Shutdown successful.
2019-11-26 20:02:32 [Main] Logging initialized successfully.
2019-11-26 20:02:32 Using settings file: /home/kochie/.config/rsudp/rsudp_settings.json
2019-11-26 20:02:32 Output directory is: /var/www/html/R2DE5
2019-11-26 20:02:32 [Init] Initializing.
2019-11-26 20:02:32 [openSOCK] Opening socket on localhost:8888 (HOST:PORT)
2019-11-26 20:02:32 [Init] Waiting for UDP data on port 8888...
2019-11-26 20:02:33 [Init] Available channels: ['EHZ', 'ENZ', 'ENN', 'ENE']
2019-11-26 20:02:33 [get_inventory] Fetching inventory for station AM.R2DE5 from Raspberry Shake FDSN.
2019-11-26 20:02:35 [get_inventory] Inventory fetch successful.
2019-11-26 20:02:35 [Write] Starting.
2019-11-26 20:02:35 [Plot] Starting.
2019-11-26 20:02:35 [Plot] Plotting channels: ['EHZ']
2019-11-26 20:02:35 [Plot] Seismogram units are Voltage counts
2019-11-26 20:02:35 [Alert] Alert stream units are Voltage counts
2019-11-26 20:02:35 [Alert] Starting Alert trigger with sta=6s, lta=30s, and threshold=1.7 on channel=?HZ
2019-11-26 20:02:35 [Consumer] Starting.
2019-11-26 20:02:35 [Producer] Starting.
2019-11-26 20:02:35 [Producer] Receiving UDP data from 10.147.17.88
2019-11-26 20:02:35 [Write] miniSEED output directory: /var/www/html/R2DE5/data
2019-11-26 20:02:35 [Write] Writing inventory to output directory.
2019-11-26 20:02:35 [Alert] Listening to channel EHZ
2019-11-26 20:02:35 [Alert] Earthquake trigger warmup time of 30 seconds...
2019-11-26 20:02:35 [Write] Beginning miniSEED output.
2019-11-26 20:03:06 [Alert] Earthquake trigger up and running normally.

andrewkochie, 2019-11-26:
Still playing around with it. Disabled plotting and was just writing the UDP stream to a seedlink file for EHZ. Noticed that it will write all channels regardless of whether or not I only stipulate EHZ. 

This is one of the errors in the terminal while listening to the UDP packets. 2019-11-26 21:15:52 [Alert] Threshold: 1.7; Current max STA/LTA: 1.5666Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/kochie/anaconda3/envs/rsudp/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/home/kochie/anaconda3/envs/rsudp/lib/python3.7/site-packages/rsudp/c_write.py", line 132, in run
    self.write()
  File "/home/kochie/anaconda3/envs/rsudp/lib/python3.7/site-packages/rsudp/c_write.py", line 89, in write
    t.write(fh, format='MSEED', encoding=enc)
  File "/home/kochie/anaconda3/envs/rsudp/lib/python3.7/site-packages/obspy/core/trace.py", line 949, in write
    Stream([self]).write(filename, format, **kwargs)
  File "/home/kochie/anaconda3/envs/rsudp/lib/python3.7/site-packages/obspy/core/stream.py", line 1426, in write
    raise NotImplementedError(msg)
NotImplementedError: Masked array writing is not supported. You can use np.array.filled() to convert the masked array to a normal array.

2019-11-26 21:18:00 [Alert] Threshold: 1.7; Current max STA/LTA: 1.5695

iannesbitt, 2019-11-27:
I made some changes, can you do `git pull` from inside the directory, then run the install script and see if that fixes at least the error you had with the write module?

I remember not writing code to allow specific channels to be written because I didn't think anyone would want to save just one channel, so I don't know why I put the option in the settings. But I will add that functionality.

I'm still not 100% sure why the plot is freezing. Does it only happen when only the plot is enabled? Are you sure your internet connection is stable? The plot will appear to freeze when there is no data flowing in.

iannesbitt, 2019-11-27:
Just pushed functionality to write specific channels, using the syntax

```
"write": {
    "enabled": false,
    "channels": ["EHZ", "ENZ"]},
```

andrewkochie, 2019-11-27:
reinstalled everything and this morning got this again:
2019-11-27 13:07:50 [Alert] Threshold: 1.7; Current max STA/LTA: 1.5707Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/kochie/anaconda3/envs/rsudp/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/home/kochie/anaconda3/envs/rsudp/lib/python3.7/site-packages/rsudp/c_write.py", line 132, in run
    self.write()
  File "/home/kochie/anaconda3/envs/rsudp/lib/python3.7/site-packages/rsudp/c_write.py", line 94, in write
    t.write(outfile, format='MSEED', encoding=enc)
  File "/home/kochie/anaconda3/envs/rsudp/lib/python3.7/site-packages/obspy/core/trace.py", line 949, in write
    Stream([self]).write(filename, format, **kwargs)
  File "/home/kochie/anaconda3/envs/rsudp/lib/python3.7/site-packages/obspy/core/stream.py", line 1426, in write
    raise NotImplementedError(msg)
NotImplementedError: Masked array writing is not supported. You can use np.array.filled() to convert the masked array to a normal array.

2019-11-27 13:08:21 [Alert] Threshold: 1.7; Current max STA/LTA: 1.5698^C

andrewkochie, 2019-11-27:
Tried cloning fresh copy again to just default settings and python throws the same error after some time.  Not sure on the masked array error

iannesbitt, 2019-11-27:
Ok try the commit I just made. I made a more explicit effort to convert from a masked array to a regular array with zeros filling the gaps.

andrewkochie, 2019-11-27:
double check repodata.json`Found conda-forge channel in /home/oem/.condarc
Creating and installing rsudp conda environment...
Collecting package metadata (current_repodata.json): failed

UnavailableInvalidChannel: The channel is not accessible or is invalid.
  channel name: xonda-forge
  channel url: https://conda.anaconda.org/xonda-forge
  error code: 404

`

iannesbitt, 2019-11-27:
That is a very strange error. Nowhere in my code is there anything that tells it to download from the "xonda-forge" channel. It should be "conda-forge". So I'm not sure why this is happening, unless there's an error in Anaconda's code, or you changed your `~/.condarc` file.

For now what you can do instead is the following (remember to replace `<your rsudp directory` with the appropriate path).
If you're on Mac or Linux:
```bash
cd <your rsudp directory>
git pull
bash unix-start-rsudp.sh
```

If you're on Windows in an Anaconda Prompt:
```
cd <your rsudp directory>
git pull
conda activate rsudp
pip install <your rsudp directory>
rs-client
```

andrewkochie, 2019-11-27:
I apologize that was my error. I ended up adding a wrong channel. Git cloned again and has been running for 30 min or so with no errors with 2 events detected. I'll keep an eye on it to see if there are any bugs.

iannesbitt, 2019-11-28:
Fantastic, great to hear! Closing this issue, if you have any further related issues feel free to reopen.

closed 2019-11-28 by iannesbitt