raspishake/rsudp #2: C_Write not writing seedlink

andrewkochie, 2019-11-30:
Error on rsudp launch is new and will not write data to seedlink file

2019-11-30 19:23:29 [Producer] Receiving UDP data from 10.147.17.88
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/home/oem/miniconda3/envs/rsudp/lib/python3.7/threading.py", line 926, in _bootstrap_inner
    self.run()
  File "/home/oem/miniconda3/envs/rsudp/lib/python3.7/site-packages/rsudp/c_write.py", line 111, in run
    self.elapse()
  File "/home/oem/miniconda3/envs/rsudp/lib/python3.7/site-packages/rsudp/c_write.py", line 75, in elapse
    self.newday = UTCDateTime(self.y, self.m, self.d + 1, 0, 0)
  File "/home/oem/miniconda3/envs/rsudp/lib/python3.7/site-packages/obspy/core/utcdatetime.py", line 359, in __init__
    dt = datetime.datetime(*args, **kwargs)
ValueError: day is out of range for month

is it possible to have one python script that reads the UDP stream and writes the miniseed files without having to plot the rest?

iannesbitt, 2019-12-01:
Yes, this error was also reported here and is due to a careless error on my part: https://community.raspberryshake.org/t/rsudp-v0-4-0-released/787/17?u=iannesbitt

You can turn off the plot by changing
```
"plot": {
    "enabled": true,
```
to
```
"plot": {
    "enabled": false,
```

andrewkochie, 2019-12-01:
Cool deal appreciate your time.

iannesbitt, 2019-12-02:
I'm closing this for now but if there are any issues let me know. I genuinely have no idea what will happen when a leap second is added.

closed 2019-12-02 by iannesbitt