raspishake/rsudp #9: Feature Request - Additional Twitter Hashtags

kipdee, 2020-10-17:
Hi,

Suggestion, on the twitter module, the ability to add other hashtags from the configuration file.  For example In my case I would add #earthquake #sismo #USVIWX #USVI

(#RaspberryShake station AM.R7078 - region: Virgin Islands) Event detected at 2020-10-16 05:52:08.20 UTC - live feed 
 https://raspberryshake.net/stationview/#?net=AM&sta=R7078 <inserted here>

(#RaspberryShake station AM.R7078 - region: Virgin Islands) Image of event detected at 2020-10-16 05:52:08.20 UTC <inserted here>

Thanks

iannesbitt, 2020-12-30:
Hi @kipdee, I think it would be reasonable to add this to the config file. It might take a little while though, since I don't work on this full time anymore.

iannesbitt, 2021-01-08:
The [extra-text](https://github.com/raspishake/rsudp/tree/extra-text) branch will contain this feature until it can be fully tested.

kipdee, 2021-01-09:
Hi Ian,

  Its Hector,  I'm running it on @QuakeUsvi and its working.  If I see any issues I'll let you know

Thanks

This was a stump test :) 
[image]

iannesbitt, 2021-01-15:
Great to hear. This feature is now merged to master.
Documentation for this feature can be found at https://raspishake.github.io/rsudp/settings.html#tweets-twitter-notification-module

closed 2021-01-15 by iannesbitt