raspishake/rsudp #13: Some minor remarks on the documentation

fwalter, 2020-12-30:
Hi again,

MINOR TYPOS IN DOCUMENTATION AND CODE MESSAGES:

"You canthen  enter the rsudp conda environment by typing "conda activate rsudp" and then run rsudp by using the command "rs-client -h""

"The client then instantiates the relevant threading.Thread objects as and passes"

In the configuration file and documentation, the sound setting is sometimes referred to as "alarm sound" and sometimes as "alert sound". I assume these are the same?

REMARKS ON DOCUMENTATION

6. Troubleshooting rsudp: "Once you have a good setup worked out, you will typically be able to start and run this program without errors.": This is a somewhat trivial statement and may annoy users who are not getting their Raspberry devices running. I suggest being more specific here or removing this sentence.
 
1.2.1. Message types: why not document "data"? As a non-python expert (though frequent user) I spent some time googling what the data types starting with a "b" mean and how they have to be handled. Then I had to find the correct methods in the rsudp modules. It would be easier to provide a function that does all this for users only interested in the data stream (via an obspy stream object, e.g., see my other issue [https://github.com/raspishake/rsudp/issues/12](url) ).

[https://github.com/openjournals/joss-reviews/issues/2565](url)

fwalter, 2020-12-30:
One more comment: When changing the python modules I noticed that the changed code is not executed unless the start-up script is re-run. Again, I am not a python expert, so this may be obvious, but it would have helped me to see this in the documentation.

iannesbitt, 2020-12-30:
These are very helpful fixes, thank you. I will work on incorporating them.

iannesbitt, 2021-01-19:
These suggestions have been implemented:

- https://github.com/raspishake/rsudp/blob/8983ce7d74aad3d1508e42119b25d9bdc5dcd62f/bin/unix-install.sh#L269
- https://raspishake.github.io/rsudp/theory.html
- https://github.com/raspishake/rsudp#rsudp
- https://raspishake.github.io/rsudp/troubleshooting.html
- https://raspishake.github.io/rsudp/settings.html

closed 2021-01-19 by iannesbitt