raspishake/rsudp #19: No rsudp_settings.json file appears.

bschellhaas, 2021-02-22:
I have unzipped the software, ran the .bat file, however, no files appear in the .config/rsudp. There didn't seem to be any errors in the install prompt, however, when I tried to run the program, this error occurred. I have also tried the advanced install, however, the same outcome happens. Any advice would be appreciated.
[image]

bschellhaas, 2021-02-22:
I tried recreating the .json file found in the install tutorial, but the same FileNotFoundError happened. Still need help.

iannesbitt, 2021-03-05:
Hi @bschellhaas:
What happens when you run the command `rs-client -i`?

alihanks, 2022-07-15:
I'm having the same issue. When trying to perform a new installation of rsudp, it states that the rsudb_settings.json file is not found. I don't see an example .json file anywhere in the installation tutorial anymore, either.

kylebrill, 2022-10-03:
I can confirm that when installing using directions in [2.2.2](https://raspishake.github.io/rsudp/installing.html#advanced-users) under the Windows install instructions, that navigating to my Miniconda environment directory: `%USERPROFILE%\Miniconda3\envs\rsudp\`, going into the `Scripts` directory, and running `rs-client.exe -i` did create the **rsubd_settings.json** file in my `%USERPROFILE%\.config\rsudp` directory.
Running `rs-settings.exe` in the same Miniconda environment directory opened an editor to work with the json file.
Might want to update the instructions now that the "Easy Way" doesn't seem to be working anymore?
[Download](https://github.com/raspishake/rsudp/releases/download/1.1.0/rsudp.zip) results in a `Not Found`.

iannesbitt, 2022-10-04:
Hi @kylebrill, I've corrected the link endpoint for [Download](https://github.com/raspishake/rsudp/archive/refs/heads/master.zip). Sorry, this was broken by a github code change that I did not pick up on. I added manual instructions in case it breaks again.

I don't do the development for the Windows scripts, so I will have to look into why this is happening and let our Windows guy know. Thanks for testing, good to know that the `-i` flag still works at least.

dmitryint, 2024-05-28:
We will close this issue but feel free to re-open it if you think that we missed something.

closed 2024-05-28 by dmitryint