raspishake/rsudp #34: Feature request: STA/LTA value plot with trigger lines

Larusrafn, 2021-12-01:
Is there a possibility to add a new optional graph section to rsudp graph window that plots the sta/lta value and displays the trigger lines? my preference would be to have it above the EHZ channel in the graphing window.....(top most graph)

similar to the lower graph in "12.3.1. Classic Sta Lta" on this page: https://docs.obspy.org/tutorial/code_snippets/trigger_tutorial.html

this would greatly help newbies like myself understand the triggering better.

Larusrafn, 2021-12-03:
minimalist suggestion: a mini version of the plot with only ehz and said sta lta plot in a new window? as a separate consumer?

iannesbitt, 2021-12-09:
Hi @Larusrafn, I have considered this and I think it would be very helpful. It's definitely on my list of things to experiment with in a future release. Thank you for the suggestion!

Larusrafn, 2021-12-12:
hope im not intruding, but the more i think about this, the more i think it should be a separate window from the main... plotting only the alert version of the stream and the sta/lta info... either on one plot with 2 scales or 2 plots. This is because you bandpass for the the alert trigger.... it would also help to see the signal difference between the main one and bandpassed. but this is maybe too much..... i am trying to hack this into the c_alert.py but as i am not a programmer, this is very slow progress :)

Lallirafn, 2022-12-06:
Anything been done about this request?

iannesbitt, 2022-12-06:
I have not done any work on this yet, but I plan on doing some work on the plotting module soon, so this will likely get included in that work.

Lallirafn, 2025-01-14:
any hope this will be implemented soon? @dmitryint?

dmitryint, 2025-02-25:
To implement this, the `Plot` class has been redesigned. There is now a `PlotController` class responsible for managing multiple instances of the `Plot` class and its descendants. The `Plot` class itself has been given a parent class, `AbcPlot`, and an additional class, `PlotAlert`, has been introduced to handle the rendering of trigger activation lines. The data for rendering is sourced from a queue based on events marked as `ALARM` and `RESET`. Queue handling has been completely moved to `PlotController`, which receives messages from the queue and forwards them to `Plot` instances.  

This is a significant change. Previously, the `Plot` class was designed to run as a separate thread. This is no longer possible. While it is now feasible to adapt `PlotController` to run in a separate thread, I see no benefit in doing so due to potential rendering issues with the `matplotlib` library. There may still be artifacts or references in the code and documentation that describe `Plot` as a separate thread. We will need to address this in the future, but I don’t expect it to cause major issues.  

The alarm trigger visualization is implemented in two possible ways, controlled by the `on_plot` parameter in the `alert` settings section. It can take three possible values:  
- `False` – disabled  
- `separate` – displayed in a separate window  
- `on-main` – displayed in the main window  

Additionally, I have added configuration options for customizing the colors of the displayed lines: `on_plot_end_line_color` and `on_plot_start_line_color`.  

Note: I suspect that performance issues may arise during prolonged operation, as there is no cleanup mechanism for lists storing line data. I haven't had the opportunity to test this yet, but I expect it will be addressed in a future update.

Lallirafn, 2025-03-08:
Thank you for this, its a superb addition to see the trigger points, but my main question was about the lower graph, that would essentialy plot the STA-LTA value under the graph of the waveform, it would really help understand the trigger mechanism and fine tune the triggering (it could be either superimposed onto the graph of the waveform or have its own between the waveform and the frequency graph. 

[image]

maybe something similar to this: 

[image]

Lallirafn, 2025-04-23:
Any hope for something like this in the near future?

Lallirafn, 2026-11-10:
still waiting for this feature.... :)