raspishake/rsudp #36: Feature request: bandpass-filtering the time series

shicks-seismo, 2022-01-01:
Hi @iannesbitt, finally got round to trying rsudp - great work!

I was wondering whether it would be possible to add an option to filter the waveforms before they are plotted live? For example, it would be a nice usage to have 2 rsudp windows open - one that is ready for teleseismic earthquakes at low freqs; the other for local detections at high freqs :)

iannesbitt, 2022-01-03:
Hi Stephen, I'm glad you enjoy it, that means a lot!

Yes, I have been considering doing this for some time now. I'm not exactly sure the difficulty of this—I have a vague memory of trying to implement it when I was first developing this but I didn't know how to set the spectrogram frequency limits to match (i.e. the bandpass worked but changing the frequency limits and plot spacing was too hard to figure out at the time).

iannesbitt, 2022-01-03:
Note: since this is changing how the plot display works, this might be good to implement in tandem with #8.

shicks-seismo, 2022-01-05:
Yeah for me, it doesn't really matter if the spectrogram is still for the raw data and remains plotting all frequencies. If so, just maybe needs a small marker line showing the bandpass frequencies. I can't remember how this looks in SeisComp3, but that might be a good model to follow.

iannesbitt, 2022-01-06:
The lines are a good idea. I was thinking I'd have to modify the extents of the raster that got plotted. I will check to see how SeisComp3 does it.

sbcaron, 2022-04-27:
Hi @iannesbitt,
I second this request :). I think I mentioned adding this feature in the RShake discussion board some time ago. I also do not see a need to filter the spectro. That seems to work just fine for me unfiltered. The real need is to filter out the noise to view the smaller and distant quakes. 

Thanks,
Steve Caron
RD29A Chino Hills.

closed 2025-01-16 by dmitryint