raspishake/rsudp #37: Install fails

PhilipPeake, 2022-02-02:
Running the unix install script from the "Latest" link on githum, on an Almalinux (centos 8 clone) it fails thus:

....
Appending sourcing line to bashrc...
Sourcing...
Activating conda...
No /home/rsudp/.condarc file exists. Creating...
Found conda-forge channel in /home/rsudp/.condarc
Creating and installing rsudp conda environment...
Collecting package metadata (current_repodata.json): done
Solving environment: failed with repodata from current_repodata.json, will retry with next repodata source.
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - matplotlib'[version='<3.2']

Current channels:

  - https://repo.anaconda.com/pkgs/main/linux-64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/linux-64
  - https://repo.anaconda.com/pkgs/r/noarch
  - https://conda.anaconda.org/rpi/linux-64
  - https://conda.anaconda.org/rpi/noarch
  - https://conda.anaconda.org/conda-forge/linux-64
  - https://conda.anaconda.org/conda-forge/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

ERROR: rsudp failed to install.
---------------------------------
Something went wrong.
Check the error output and try again.

tian3rd, 2022-02-28:
I have the same error message. Mine is Raspbian GNU/Linux 11 (bullseye).

maniflames, 2022-03-17:
Same issue here, the installer fails but if you create the conda env manually everything installs and runs as it should. 

```bash
conda env create
conda activate rsudp
./unix-start-rsudp.sh
```

Note that the datacast of the shake has to be active before the script can successfully visualize something.

cappyccino, 2022-04-08:
Looks like the change from commit [64576f7](https://github.com/raspishake/rsudp/commit/64576f7ebe272a00a12dc889cc1cc6ba85b3c117) might be the issue. I'm not super familiar with python-- maybe @iannesbitt can take a look?

https://github.com/raspishake/rsudp/issues/21

cappyccino, 2022-04-08:
You can fix this issue by hardcoding the version to `matplotlib=3.1.3`. See my proposed changes in the commit ☝🏼

iannesbitt, 2022-04-08:
@cappyccino Thank you for your work on this, I haven't had much time to maintain lately. Since Matplotlib has so many dependencies and intricacies, new versions can break downstream software like rsudp. For now, pinning the version at `3.1.3` seems like a good solution. Do you want to create a pull request with your commit?

iannesbitt, 2022-06-16:
Just a note about matplotlib in the install script:

I have removed the installation of matplotlib from the `conda install` commands in the script and leave it to be installed via pip now.

It seems like occasionally MPL does not get built for all architectures in the conda-forge channel, and the rpi channel's matplotlib has not been built for several years' worth of versions. This was causing issues with certain architectures not finding the versions they needed when using the install script. Therefore it's probably better to have pip/setuptools build MPL during install (even though it takes longer).

Because of this I have removed both `matplotlib` and `obspy` from the initial conda environment creation step. This shouldn't cause any issues down the line but it's hard to tell for sure. I've tested the new version of the install script on `armv7l` running Debian and `x86_64` running both Ubuntu and Windows and it seems to be working ok.

This should also solve #38.

closed 2022-06-16 by iannesbitt