raspishake/rsudp #39: Refresh interval matplotlib

rankinstudio, 2022-03-20:
I have rsudp running on a PC. It works great but using matplotlib is very cpu intensive. It would be nice to have a refresh interval in the settings file for the plot so it didn't refresh as fast as it possibly can. I'd be fine with a 5s refresh if it didn't hit the cpu as hard. 

Cheers

iannesbitt, 2022-04-08:
Hi @rankinstudio, I think this should be fairly simple to implement, but I don't have time to fix it at the moment. The work required would be:
- add a refresh interval parameter in the settings file (`refresh_sec` or something similar perhaps)
- add a way to load this parameter in `client.py` and send it to the `Plot` class
- add a variable to keep track of this interval in `Plot` initialization
- use this variable to calculate the number of skipped refreshes (currently based, I think, on the number of received data packets versus the packet transmission rate)

rankinstudio, 2022-04-16:
I can take a look at it.

rankinstudio, 2025-03-23:
Thank you Dmitry!