raspishake/rsudp #41: sta/lta timing alarm

rankinstudio, 2022-04-16:
It seems, after messing with this for a while, that setting an alarm only on the threshold value being hit isn't the best approach. When there is a short lived disturbance the threshold can be very briefly hit and drop back down quickly, still sending an alert out. Earthquakes just distant / weak enough to not set the alarm off will still make the sta/lta ratio climb for a longer period of time. It would be neat to be able to configure a trigger with a lower threshold that times how long the alarm is active. If this time value is met (say 10s active) then it sends an alert out. The reset even would stop and reset the timer. I think this would be more likely to skip short events and catch earthquakes. 

Cheers

David

iannesbitt, 2022-10-04:
Hi David, great point here. The reason I originally built it this way was because I wanted our classroom demonstrations to be able to trigger the alarm just by stomping, but I totally agree with you. The fix I am envisioning would have some sort of "duration" setting which when turned above zero would require a certain number of seconds to elapse over the threshold to trip the alarm.

iannesbitt, 2022-10-04:
The things that would have to change are

- [ ] add a `"duration"` setting to the `alert` dictionary in the `.json` file. (should be a float with a default of 0.0 so classrooms don't have to reconfigure)
- [ ] add `"duration"` to `client.py` intake section
- [ ] add `"duration"` to `client.py` thread initialization section
- [ ] add `duration` as a parameter to rsudp.c_alert.Alert.__init__()
- [ ] add `duration` as a definition to rsudp.c_alert.Alert.__init__()
- [ ] change line 240 in rsudp.c_alert.Alert._is_trigger() to a separate exceedence test function
- [ ] add test function rsudp.c_alert.Alert._trigger_test() which switches test method based on duration
  - [ ] if 0 or less: self.stalta.max(); if greater than 0: would compare last `n` samples within `duration` range (`n = rsudp.raspberryshake.getSR() * duration`) and ALL would have to exceed `threshold`
- [ ] add to docstrings: `c_alert.Alert`
- [ ] add to docs: `settings.rst`

I may be forgetting a step but this should cover the important changes at least.

rankinstudio, 2024-02-13:
I added this to rsudp.c_alert.Alert.init()
self.timerRunning = False

Then this to rsudp.c_alert.Alert._is_trigger()
		'''
		Figures out it there's a trigger active.
		'''
		#PRE TRIGGER TIMER
		if self.stalta.max() > self.thresh:
			if not self.exceed: #Not in triggered state already
				if not self.timerRunning:
					self.t = threading.Timer(5, lambda:self.timerStartedMsg())
					self.t.start()
					self.timerRunning = True

				if self.timerRunning:
					if self.t.is_alive():
						printM("Timer running")
					else:
						printM("Timer dead")
						self.timerRunning = False
						self.t.cancel()
		else:
			if self.timerRunning:
				printM("Timer canceled")
				self.t.cancel()
				self.timerRunning = False

		#TRIGGER PASSED TIMER - continue to regular trigger.
		if self.stalta.max() > self.thresh and not self.timerRunning: #add check here for timer running
			if not self.exceed:
                        etc....

I don't know how to implement this fully the way you wanted to, but I've tested it extensively and it works well. 

Cheers
David

rankinstudio, 2024-02-13:
I added this to rsudp.c_alert.Alert.init()
`self.timerRunning = False`

Then this to rsudp.c_alert.Alert._is_trigger()

	def _is_trigger(self):
		'''
		Figures out it there's a trigger active.
		'''
		#PRE TRIGGER TIMER
		if self.stalta.max() > self.thresh:
			if not self.exceed: #Not in triggered state already
				if not self.timerRunning:
					self.t = threading.Timer(5, lambda:self.timerStartedMsg())
					self.t.start()
					self.timerRunning = True

				if self.timerRunning:
					if self.t.is_alive():
						printM("Timer running")
					else:
						printM("Timer dead")
						self.timerRunning = False
						self.t.cancel()
		else:
			if self.timerRunning:
				printM("Timer canceled")
				self.t.cancel()
				self.timerRunning = False

		#TRIGGER PASSED TIMER
		if self.stalta.max() > self.thresh and not self.timerRunning: #added timer running check here
			if not self.exceed:
                        etc.

I don't know how to implement this fully the way you wanted to, but I've tested it extensively and it works well. 

Cheers
David

rankinstudio, 2024-03-01:
Just want to say that after running the above code for a while, it has decreased the false alarms to almost non-existent, and is extremely good at capturing quakes, even muted ones.