raspishake/rsudp #42: rs-test fails, Import-Error

tobidurr, 2022-04-25:
I installed rsudp on two separate machines, one running Win10, one Win11. As far as I can see the installation works successfully. When I try to run rs-test in the rsudp environment I get the following error message (shortened) on both machines. 
```
 PS C:\Users\Abberior_Service> rs-test
Traceback (most recent call last):
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\site-packages\obspy\core\util\libnames.py", line 98, in _load_cdll
    cdll = ctypes.CDLL(libpath)
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\ctypes\__init__.py", line 364, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: [WinError 126] The specified module could not be found

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\Scripts\rs-test.exe\__main__.py", line 4, in <module>
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\site-packages\rsudp\client.py", line 20, in <module>
    from rsudp.c_alert import Alert
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\site-packages\rsudp\c_alert.py", line 4, in <module>
    from obspy.signal.trigger import recursive_sta_lta, trigger_onset
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\site-packages\obspy\signal\__init__.py", line 201, in <module>
    from .spectral_estimation import PPSD
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\site-packages\obspy\signal\spectral_estimation.py", line 47, in <module>
    from obspy.signal.invsim import cosine_taper
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\site-packages\obspy\signal\invsim.py", line 38, in <module>
    from obspy.signal import util
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\site-packages\obspy\signal\util.py", line 24, in <module>
    from obspy.signal.headers import clibsignal
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\site-packages\obspy\signal\headers.py", line 18, in <module>
    clibsignal = _load_cdll("signal")
  File "C:\Users\Abberior_Service\.conda\envs\rsudp\lib\site-packages\obspy\core\util\libnames.py", line 112, in _load_cdll
    raise ImportError(msg)
ImportError: Could not load shared library "libsignal_Windows_64bit_py37.pyd"
  Path: C:\Users\Abberior_Service\.conda\envs\rsudp\lib\site-packages\obspy\lib\libsignal_Windows_64bit_py37.pyd
  Current directory: C:\Users\Abberior_Service
  ctypes error message: [WinError 126] The specified module could not be found
  Directory listing of lib directory:
    -, ., :, A, C, S, U, \, \, \, \, \, \, \, \, \, _, a, a, a, b, b, b,
    b, b, c, c, c, d, d, e, e, e, e, e, e, e, g, i, i, i, i, i, k, l, l,
    n, n, o, o, o, p, p, p, r, r, r, r, r, s, s, s, s, s, s, s, t, u, v,
    v, y
```

I am not sure if this is an issue with rsudp but googling the error code did not help. I am therefore not able to run rsudp currently.

rodonile, 2022-06-21:
I'm also having this same issue.

iannesbitt, 2022-10-04:
> `ImportError: Could not load shared library "libsignal_Windows_64bit_py37.pyd"`

@tobidurr @rodonile @Jos3111 Thank you for the reports.

This indeed appears to be an upstream issue with an obspy dependency. How did you install obspy and what version are you using? Also, what version of Windows?
If you don't want to use the workaround of commenting out the obspy dependencies, consider [opening an issue with obspy](https://github.com/obspy/obspy/issues/new/choose). It looks like they've had [a similar issue to this](https://github.com/obspy/obspy/pull/1636) in the past.

dmitryint, 2024-05-28:
We will close this issue but feel free to re-open it if you think that we missed something.