raspishake/rsudp #44: Installing on Ubuntu18.04

hadigh, 2022-08-09:
Hi 

I am trying to install rsudp on a Linux machine (18.04.1-Ubuntu) by running "bash unix-install-rsudp.sh"; I am getting the below error and was wondering if you can help to solve the issue! my Conda version is: conda 4.13.0

Solving environment: \ 
Found conflicts! Looking for incompatible packages.
This can take several minutes.  Press CTRL-C to abort.
failed                                                                                                                                                                                                      

UnsatisfiableError: The following specifications were found to be incompatible with each other:

Output in format: Requested package -> Available versions

Package expat conflicts for:
scipy -> pypy3.8[version='>=7.3.9'] -> expat[version='>=2.2.9,<3.0.0a0|>=2.3.0,<3.0a0|>=2.4.1,<3.0a0|>=2.4.8,<3.0a0']
lxml -> pypy3.9[version='>=7.3.9'] -> expat[version='>=2.2.9,<3.0.0a0|>=2.4.1,<3.0a0|>=2.4.8,<3.0a0|>=2.3.0,<3.0a0']
future -> pypy3.9[version='>=7.3.8'] -> expat[version='>=2.2.9,<3.0.0a0|>=2.3.0,<3.0a0|>=2.4.1,<3.0a0|>=2.4.7,<3.0a0|>=2.4.8,<3.0a0']
sqlalchemy -> pypy3.9[version='>=7.3.9'] -> expat[version='>=2.2.9,<3.0.0a0|>=2.3.0,<3.0a0|>=2.4.1,<3.0a0|>=2.4.7,<3.0a0|>=2.4.8,<3.0a0']
cryptography -> pypy3.8[version='>=7.3.9'] -> expat[version='>=2.2.9,<3.0.0a0|>=2.4.1,<3.0a0|>=2.4.7,<3.0a0|>=2.4.8,<3.0a0|>=2.3.0,<3.0a0']

Package python_abi conflicts for:
scipy -> python_abi[version='3.10.*|3.9.*|3.8.*|3.8|3.9|3.7|3.7.*|3.6.*|3.6',build='*_pypy36_pp73|*_pypy37_pp73|*_cp310|*_cp39|*_cp38|*_pypy38_pp73|*_pypy39_pp73|*_cp37m|*_cp36m']
lxml -> python_abi[version='2.7.*|3.10.*|3.7.*|3.8.*|3.9.*|3.9|3.8|3.7|3.6.*|3.6',build='*_pypy36_pp73|*_cp36m|*_pypy37_pp73|*_pypy39_pp73|*_cp38|*_cp37m|*_cp39|*_cp310|*_pypy38_pp73|*_cp27mu']
cryptography -> python_abi[version='2.7.*|3.10.*|3.7.*|3.8.*|3.8|3.9|3.9.*|3.7|3.6.*|3.6',build='*_cp27mu|*_cp36m|*_pypy37_pp73|*_pypy39_pp73|*_cp310|*_cp38|*_cp37m|*_pypy38_pp73|*_cp39|*_pypy36_pp73']
future -> python_abi[version='2.7.*|3.10.*|3.7.*|3.9|3.8.*|3.9.*|3.8|3.7|3.6.*|3.6',build='*_pypy39_pp73|*_cp37m|*_cp38|*_cp310|*_cp39|*_pypy38_pp73|*_pypy37_pp73|*_cp36m|*_pypy36_pp73|*_cp27mu']
numpy=1.16.4 -> python[version='>=3.6,<3.7.0a0'] -> python_abi[version='3.10.*|3.6|3.7|3.8.*|3.7.*|3.9|3.9.*|3.8|3.6.*',build='*_cp36m|*_pypy37_pp73|*_pypy36_pp73|*_cp38|*_cp37m|*_pypy39_pp73|*_cp39|*_pypy38_pp73|*_cp310']
sqlalchemy -> python_abi[version='2.7.*|3.10.*|3.7.*|3.9|3.8|3.9.*|3.8.*|3.7|3.6.*|3.6',build='*_pypy36_pp73|*_pypy37_pp73|*_cp310|*_pypy39_pp73|*_cp37m|*_pypy38_pp73|*_cp39|*_cp38|*_cp36m|*_cp27mu']

Package numpy conflicts for:
scipy -> numpy[version='1.10.*|1.11.*|1.12.*|1.13.*|>=1.11|>=1.11.3,<2.0a0|>=1.14.6,<2.0a0|>=1.16.5,<2.0a0|>=1.16.6,<2.0a0|>=1.17.5,<2.0a0|>=1.18.5,<2.0a0|>=1.19.5,<2.0a0|>=1.21.6,<2.0a0|>=1.21.5,<2.0a0|>=1.21.4,<2.0a0|>=1.19.4,<2.0a0|>=1.19.2,<2.0a0|>=1.18.1,<2.0a0|>=1.9.3,<2.0a0|>=1.9|>=1.21.2,<2.0a0|>=1.15.1,<2.0a0']
numpy=1.16.4

Package python conflicts for:
numpy=1.16.4 -> mkl_fft[version='>=1.0.6,<2.0a0'] -> python[version='2.7.*|3.5.*|3.6.*|>=3.10,<3.11.0a0|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0|>=3.5,<3.6.0a0']
future -> python[version='2.7.*|3.4.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.10,<3.11.0a0|>=3.7,<3.8.0a0|>=3.9,<3.10.0a0|>=3.8,<3.9.0a0|>=3.6,<3.7.0a0|>=3.5,<3.6.0a0']
scipy -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.10,<3.11.0a0|>=3.9,<3.10.0a0|>=3.8,<3.9.0a0|>=3.7,<3.8.0a0|>=3.6,<3.7.0a0|>=3.5,<3.6.0a0|3.4.*']
sqlalchemy -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.10,<3.11.0a0|>=3.7,<3.8.0a0|>=3.9,<3.10.0a0|>=3.8,<3.9.0a0|>=3.6,<3.7.0a0|>=3.5,<3.6.0a0|3.4.*']
numpy=1.16.4 -> python[version='>=2.7,<2.8.0a0|>=3.6,<3.7.0a0|>=3.7,<3.8.0a0']
future -> python_abi=3.7[build=*_cp37m] -> python[version='3.10.*|3.7.*|3.8.*|3.9.*']
lxml -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.10,<3.11.0a0|>=3.7,<3.8.0a0|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0|>=3.6,<3.7.0a0|>=3.5,<3.6.0a0|3.4.*']
scipy -> python_abi=3.9[build=*_cp39] -> python[version='3.10.*|3.9.*|3.8.*|3.7.*']
python=3.7.4
cryptography -> cffi[version='>=1.12'] -> python[version='3.10.*|3.6.12|3.6.12|3.7.10|3.7.10|3.7.12|3.8.12|3.9.10|3.7.9|3.6.9|3.6.9|3.6.9|3.6.9|3.8.*|3.7.*|3.9.*|3.4.*|>=3.6|>=3.5|<=3.3',build='0_73_pypy|1_73_pypy|4_73_pypy|5_73_pypy|0_73_pypy|1_73_pypy|0_73_pypy|5_73_pypy|3_73_pypy|2_73_pypy']
lxml -> python_abi=3.7[build=*_cp37m] -> python[version='3.10.*|3.7.*|3.8.*|3.9.*']
sqlalchemy -> importlib-metadata -> python[version='3.10.*|3.7.*|>=3|>=3.6|3.9.*|3.8.*']
cryptography -> python[version='2.7.*|3.5.*|3.6.*|>=2.7,<2.8.0a0|>=3.10,<3.11.0a0|>=3.7,<3.8.0a0|>=3.8,<3.9.0a0|>=3.9,<3.10.0a0|>=3.6,<3.7.0a0|>=3.5,<3.6.0a0']

Package pypy3.6 conflicts for:
cryptography -> pypy3.6[version='>=7.3.1|>=7.3.2|>=7.3.3']
cryptography -> cffi[version='>=1.12'] -> pypy3.6[version='7.3.*|7.3.0.*|7.3.1.*|7.3.2.*|7.3.3.*']

Package pypy3.8 conflicts for:
cryptography -> pypy3.8[version='>=7.3.8|>=7.3.9']
cryptography -> cffi[version='>=1.12'] -> pypy3.8[version='7.3.*|7.3.8.*|7.3.9.*']

Package _openmp_mutex conflicts for:
cryptography -> libgcc-ng[version='>=12'] -> _openmp_mutex[version='>=4.5']
sqlalchemy -> libgcc-ng[version='>=12'] -> _openmp_mutex[version='>=4.5']
numpy=1.16.4 -> libgcc-ng[version='>=7.3.0'] -> _openmp_mutex[version='*|>=4.5',build=*_llvm]
lxml -> libgcc-ng[version='>=12'] -> _openmp_mutex[version='>=4.5']
python=3.7.4 -> libgcc-ng[version='>=7.3.0'] -> _openmp_mutex[version='>=4.5']
scipy -> libgcc-ng[version='>=12'] -> _openmp_mutex[version='*|>=4.5',build=*_llvm]

Package pypy3.9 conflicts for:
cryptography -> pypy3.9[version='>=7.3.8|>=7.3.9']
cryptography -> cffi[version='>=1.12'] -> pypy3.9[version='7.3.*|7.3.8.*|7.3.9.*']

Package six conflicts for:
scipy -> mkl-service[version='>=2.3.0,<3.0a0'] -> six
cryptography -> six[version='>=1.4.1']

Package pypy3.7 conflicts for:
cryptography -> cffi[version='>=1.12'] -> pypy3.7[version='7.3.*|7.3.3.*|7.3.4.*|7.3.5.*|7.3.7.*']
cryptography -> pypy3.7[version='>=7.3.3|>=7.3.5|>=7.3.7']The following specifications were found to be incompatible with your system:

  - feature:/linux-64::__glibc==2.27=0
  - feature:|@/linux-64::__glibc==2.27=0
  - cryptography -> libgcc-ng[version='>=9.3.0'] -> __glibc[version='>=2.17']
  - lxml -> libgcc-ng[version='>=9.3.0'] -> __glibc[version='>=2.17']
  - numpy=1.16.4 -> libgcc-ng[version='>=7.3.0'] -> __glibc[version='>=2.17']
  - scipy -> libgfortran-ng -> __glibc[version='>=2.17']
  - sqlalchemy -> libgcc-ng[version='>=9.3.0'] -> __glibc[version='>=2.17']

Your installed version is: 2.27

Note that strict channel priority may have removed packages required for satisfiability.

ERROR: rsudp failed to install.
---------------------------------
Something went wrong.
Check the error output and try again.

iannesbitt, 2022-10-04:
Hi @hadigh, thanks for the report. Can you post the entire command output? It looks like you're just posting the last hundred lines or so. You may have to expand the number of lines your terminal remembers in order to get to the first few.

Operating on limited information, my guess is either that your anaconda installation is outdated and needs to be reinstalled, or that you have an existing `rsudp` environment that needs to be removed before you install a new one.

In the former case, you may be able to work around these errors by updating your conda installation. You can do this by executing the following command:

```
conda update -n base conda
```

Some of the older installations don't update smoothly, so on occasion it's easier to start over than to work through all the errors with updating. If the update command fails you may need to reinstall anaconda, which requires deleting your anaconda installation folder (usually `~/anaconda3` or `~/miniconda3`. You can preserve your existing envs by moving your `~/[ana|mini]conda3/envs/` folder elsewhere before removing the base conda folder. Move `envs/` back into the base conda folder to restore the envs once the new installation is done.

navneetercse, 2022-11-01:
Seeing the error messages, there is a very high chance of an existing rsudp environment under an already installed conda.

Also, can we provide an option for overwriting of existing rsudp environment during installation (via a prompt)?

iannesbitt, 2022-11-01:
> Also, can we provide an option for overwriting of existing rsudp environment during installation (via a prompt)?

Sure, that shouldn't be too difficult. Most of the time it will just be in `~/<conda directory>/envs/rsudp`, and just needs to be `rm -rf`'d before reinstallation. I will go ahead and take a look at adding this.

iannesbitt, 2022-11-23:
Should be resolved in #45.

closed 2022-11-23 by iannesbitt