raspishake/rsudp #49: (OSX/Darwin) Install fails on Apple M1 (arm64) architecture

iannesbitt, 2022-12-03:
The rsudp installer still thinks Macs use Intel chips. Newer ones have adopted and updated the ARM architecture, I think due to power consumption improvements. Here are some things that currently fail:

- installer downloads `x86_64` by default, needs to download `arm64` for M1 machines
- pip wheel build fails (`matplotlib<3.2`, also possibly `obspy`)
  - this is actually kind of a significant problem since I think the reason we pinned the version at 3.1.3 and below is because some critical feature for display updating disappears in 3.2. I think that means stuff in `c_plot.py` needs to be rewritten to avoid depreciation errors. I need to test to confirm this. If we bump the version, I think I will need to find a workaround for those features disappearing. I remember being disappointed at the options when I discovered this issue but we'll see.

CesarSP99, 2023-08-04:
I can confirm this issue too as it seems that there is no version available for ARM based chipsets. Here I have the relevant output of the installer script:

`
Collecting package metadata (repodata.json): done
Solving environment: failed

PackagesNotFoundError: The following packages are not available from current channels:

  - numpy=1.16.4

Current channels:

  - https://conda.anaconda.org/conda-forge/osx-arm64
  - https://conda.anaconda.org/conda-forge/noarch
  - https://repo.anaconda.com/pkgs/main/osx-arm64
  - https://repo.anaconda.com/pkgs/main/noarch
  - https://repo.anaconda.com/pkgs/r/osx-arm64
  - https://repo.anaconda.com/pkgs/r/noarch

To search for alternate channels that may provide the conda package you're
looking for, navigate to

    https://anaconda.org

and use the search bar at the top of the page.

ERROR: rsudp failed to install.
---------------------------------
Something went wrong.
Check the error output and try again.
`

dmitryint, 2024-09-06:
We have updated matplotlib and obspy versions and added support for arm64 architecture.
[Eurasian](https://github.com/raspishake/rsudp/releases/tag/2.0.0)

closed 2024-09-06 by dmitryint