raspishake/rsudp #51: problem on accessing waveforms using rsudp

testaisusing, 2023-01-18:
Hii, I am a new user to raspberry pi and I want to take advantages of the  rsudp to visualize the waveforms from the one of the open station situated here https://stationview.raspberryshake.org/#/?lat=9.85918&lon=77.89304&zoom=4.847&net=AM&sta=R7186 . However when i run  'rs-client' after changing the setting file on linux command line. I get the error like as follows even if the station data is open and the waveforms can be visualized in the above link. Ip address of the laptop is  IP address: 10.10.1.203    Port: 8888. I am using wifi to access the waveforms.

I hope experts may help me organizing the setting file properly. Thanks.

2023-01-18 07:22:21 [Init] Logging initialized successfully.
2023-01-18 07:22:21 Using settings file: /home/xxxx/.config/rsudp/rsudp_settings.json
2023-01-18 07:22:21 Output directory is: /home/xxxx/rsudp
2023-01-18 07:22:21 [RS lib] Initializing rsudp v 1.1.0.
2023-01-18 07:22:21 [openSOCK] Opening socket on localhost:8888 (HOST:PORT)
2023-01-18 07:22:21 [RS lib] Waiting for UDP data on port 8888...
2023-01-18 07:22:31 [Init] ERROR: No data received in 10 seconds; aborting.
2023-01-18 07:22:31 [Init]        Check that the Shake is forwarding data to:
2023-01-18 07:22:31 [Init]        IP address: 10.10.1.203    Port: 8888
2023-01-18 07:22:31 [Init]        and that no firewall exists between the Shake and this computer.

navneetercse, 2023-03-12:
You need to forward data from the device(raspberry shake) to your computer's IP address to a particular UDP port, to view live data using rsudp.

dmitryint, 2024-05-28:
We will close this issue but feel free to re-open it if you think that we missed something.

closed 2024-05-28 by dmitryint