raspishake/rsudp #52: Certificate for fdsnws.raspberryshakedata.com is expired, causing inventory fetch to fail on startup.

Lallirafn, 2023-03-16:
Certificate for fdsnws.raspberryshakedata.com is expired, causing inventory fetch to fail on startup. This messes up deconvolve (i think its called) and i get only counts instead of mm/sec on the graphs. 

Do you have contacts at raspberryshake to ask them to fix the letsencrypt config and renew the certs?

[image]
[image]

[image]

Lallirafn, 2023-03-16:
when calling from a browser it sometimes correctly redirects to https://data.raspberryshake.org/fdsnws/station/1/query?network=AM&station=........................... maybe time to fix the url call in the source?

Lallirafn, 2023-03-16:
confirmed. but when changed to data.raspberryshake.org this warning is displayed instead:
[image]

iannesbitt, 2023-04-06:
It looks like the letsencrypt certificate for fdsnws.raspberryshakedata.com was renewed. Can you confirm this problem still exists?

Lallirafn, 2023-04-06:
the certificate has been renewed so the original error is resolved, but the url is redirected to https://data.raspberryshake.org/fdsnws/station/1/query?network=AM&station=(REMOVED)&level=resp&nodata=404&format=xml where the dataformat for the XML is 1.2 and ObsPy throws a warning: ".....\miniconda3\envs\rsudp\lib\site-packages\obspy\io\stationxml\core.py:98: UserWarning: The StationXML file has version 1.2, ObsPy can read versions (1.0, 1.1). Proceed with caution."

iannesbitt, 2023-04-06:
This is a message generated by Obspy, and it should be safe to ignore. Per obspy/obspy#3153, a fix for this has been merged into their maintenance branch and should be released soon. The schema difference should not affect rsudp's performance.

dmitryint, 2024-05-28:
We will close this issue but feel free to re-open it if you think that we missed something.

closed 2024-05-28 by dmitryint