raspishake/rsudp #53: Telegram sending failing with RuntimeWarning

Lallirafn, 2023-03-24:
after a fresh install of rsudp (using config from older setup that worked fine) Telegram sending fails with "RuntimeWarning":

`2023-03-23 23:58:05 [AlertSound] Playing alert sound...
2023-03-23 23:58:05 [Telegram id -1001674859833] Sending alert...
2023-03-23 23:58:05 [Telegram id -1001674859833] Telegram message: (Raspberry Shake station (removed) Event detected at 2023-03-23 23:58:05.19 UTC !! POSSIBLE EARTHQUAKE EVENT !! - live feed  https://stationview.raspberryshake.org/#?net=AM&sta=
C:\Users\user\miniconda3\envs\rsudp\lib\site-packages\rsudp\c_telegram.py:165: RuntimeWarning: coroutine 'Bot.send_message' was never awaited
  self._when_alarm(d)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
2023-03-23 23:58:06 [Plot] Event time: 2023-03-23 23:58:05.19 UTC2.1776[2K
2023-03-23 23:58:06 [Plot] Saving png in about 42 seconds
2023-03-23 23:58:16 [Alert] Threshold: 1.0; Current max STA/LTA: 0.5166
2023-03-23 23:58:17 [Alert] Max STA/LTA ratio reached in alarm state: 2.187
2023-03-23 23:58:17 [Alert] Earthquake trigger reset and active again at 2023-03-23 23:58:17.19
2023-03-23 23:58:17 [Producer] Alert thread has indicated alarm reset, sending RESET message to queues
2023-03-23 23:58:48 [Plot] Saving png 43 seconds after alarmLTA: 0.1023[2K
2023-03-23 23:58:49 [Plot] Saved C:\Users\user\rsudp\screenshots\REMOVED-2023-03-23-235805.png
2023-03-23 23:58:49 [Plot] Plot thread has saved an image, sending IMGPATH message to queues
2023-03-23 23:58:49 [Telegram id -xxxxxxxxxxxxx] Uploading image to Telegram C:\Users\rsudp\screenshots\REMOVED-2023-03-23-235805.png
2023-03-23 23:58:49 [Telegram id -xxxxxxxxxxxxx] Sent image
C:\Users\user\miniconda3\envs\rsudp\lib\site-packages\rsudp\c_telegram.py:168: RuntimeWarning: coroutine 'Bot.send_photo' was never awaited
  self._when_img(d)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback
2023-03-24 00:01:54 [Alert] Threshold: 1.0; Current max STA/LTA: 0.1219`

oguven, 2023-03-26:
I'm also having the same issue here.

Python version 3.9.9
OS: MacOS

`/usr/local/lib/python3.9/site-packages/rsudp/c_telegram.py:165: RuntimeWarning: coroutine 'Bot.send_message' was never awaited
  self._when_alarm(d)`

Lallirafn, 2023-04-04:
is this project going dormant?

Lallirafn, 2023-04-05:
This is the issue: 

Python telegram bot was completely redesigned on 2022-05-06, with release 20.0a0 and that code uses asyncio. 

To temporarily remedy the issue until c_telegram.py can be rewritten to call the bot with asyncio, you can install the older version in conda environment ("conda activate rsudp", then run "pip install python-telegram-bot==13.11" to install the old version of the python bot. Then run the rs-client as usual and Telegram sending should work again.

i have verified that this works with fresh install of current RSUDP through Conda.

iannesbitt, 2023-04-06:
Hi @Lallirafn, I appreciate your work here. The project is not officially going dormant, but I have been extremely busy lately and haven't had much time to keep up with issues. I will pin the bot to version 13.11 and will rework the interface for the next major release to fix the breaking change.

Lallirafn, 2023-04-06:
thanks for keep working on this project @iannesbitt. I will help as much as I can, but I´m not a programmer so my contributions will be of limited quality.... But there is increased pressure on updating various components. I managed to alter the code to use the new Telegram client, but got stuck on a known error in the pinned Python version for RSUDP that is fixed in 3.11. I guess python is pinned for the pinned version of Matplotlib, and that is pinned because of some functions missing in newer versions if i remember correctly?

iannesbitt, 2023-04-06:
@Lallirafn if you have managed to find a workaround, I am happy to review your code and incorporate changes into the project. Can you open a pull request with your proposed code changes?

dmitryint, 2024-09-06:
[Eurasian](https://github.com/raspishake/rsudp/releases/tag/2.0.0)

closed 2024-09-06 by dmitryint