raspishake/rsudp #62: URL needs updated, can't get config file.

rankinstudio, 2024-02-24:
If you're not able to get rsudp to get your config file for your station to deconvolve the data, the url under

rsudp.raspberryshake.py.getinventory()

needs to be updated to:
`url = 'https://stationview.raspberryshake.org/fdsnws/station/1/query?network=%s&station=%s&level=resp&nodata=404&format=xml' % (
				   net, stn)`

This is related to the recent shake server outages. 

Might be good to expose this in the settings.json

dmitryint, 2024-05-28:
Hello. Thank you for the feedback. We plan to implement this and add the ability to define a link through settings.json. Currently, the link is hardcoded into the code.

dmitryint, 2024-09-30:
https://github.com/raspishake/rsudp/pull/71