raspishake/rsudp #63: Divide by zero encountered in matplotlib

ajball2001, 2024-04-23:
I have an issue that throws "/home/pi/berryconda3/envs/rsudp/lib/python3.6/site-packages/matplotlib/axes/_axes.py:7747: RuntimeWarning: divide by zero encountered in log10
  Z = 10. * np.log10(spec)"
I am unsure how to proceed. I am running the software as Daemon and it isn't restarting when this error is thrown. It usually will run for around 10-20 minutes before this error is encountered but sometimes it is seconds after start, never errorless for more than 30 minutes.

dmitryint, 2024-05-28:
Hello. Thanks for your feedback. This warning appears to occur in matplotlib library. Can you tell me which matplotlib version you are using? In any case, if this does not cause the client (or daemon process) to stop, then it can be treated as a warning and ignored.

ajball2001, 2024-05-28:
Hi, I am using the version of matplotlib installed by the installer,
however, it seems that there is an error which is causing the process to
crash, however, the raspberry pi does not recognize that the daemon process
has stopped working and thus it cannot be restarted without a complete
restart. This is a raspberry pi 3B.

dmitryint, 2024-09-30:
Hi, We did a major code update that should fix this bug, but we had to switch to a new version of python, which made it impossible for us to automatically install new versions on old arm.

We used berriconda for this, but it is no longer developed. And it seems we have no alternatives for older versions of arm architecture.
https://github.com/jjhelmus/berryconda
https://github.com/jjhelmus/berryconda/issues/83

closed 2025-01-16 by dmitryint