raspishake/rsudp #65: new updates coming?

Lallirafn, 2024-05-30:
nice to see a bit of life here, as this is one of my favorite projects regarding raspberry shake. 

I was afraid that this project was dead, but now my hopes are up again..... will we see some updates in the near future?

chaunceygardiner, 2024-06-16:
It really would be nice to get an update.  I’m running an old Raspbery Pi with an old 32-bit Raspberry OS just to run rsudp.  Miniforge has conda for the Raspberry Pi 5 64-bit, but you’ve got out of date dependencies.  This is really nice software to detect earthquakes.  I sure hope you can find the time for another rev.  Or ask for help to get this done.

dmitryint, 2024-06-30:
Hello. Yes, the first thing we plan to do is to update the dependencies to the latest ones, adding support for the latest versions of python and newer versions of libraries if possible. We plan to release version 1.1.1 with the current changes, and release a new one with support for new python versions. Any suggestions and comments on this matter are welcome.

dmitryint, 2024-08-01:
We have migrated client to new Python versions (3.10, 3.11, 3.12) and updated dependencies.
https://github.com/raspishake/rsudp/pull/66

chaunceygardiner, 2024-08-02:
I gave the development branch a try.  The install script fails because it installs and attempts to run miniconda on Raspberry OS (bookworm) 64-bit (bookworm 64-bit is the recommended OS for 64-bit capable Raspberry Pis).  Miniconda is 32-bit.

chaunceygardiner, 2024-08-03:
Hooray!

I installed miniforge3 from here (https://github.com/conda-forge/miniforge/releases/latest/download/Miniforge3-Linux-aarch64.sh).

I then changed various scripts to look for miniforge3 as one of the sources of conda.

Now the install works.  And so does the execution.

The only thing missing is that the install script should install miniforge3 for raspberry os 64-bit.  I can probably work on that if you need help.

The git diff of the minor changes I made follows:

diff --git a/bin/unix-install.sh b/bin/unix-install.sh
index 9b991ac..94bc54f 100644
--- a/bin/unix-install.sh
+++ b/bin/unix-install.sh
@@ -21,6 +21,7 @@ if [[ "$arch" == "armv"* ]]; then release='berryconda3'; else release='miniconda
 # conda install location:
 prefix="$HOME/$release"         # $HOME/miniconda3 is default location
 full="$HOME/anaconda3"          # full release install location
+miniforge3="$HOME/miniforge3"   # MiniForge3
 berryconda="$HOME/berryconda3"  # berryconda install location
 miniconda="$HOME/miniconda3"    # miniconda install location
 config="$HOME/.config/rsudp"    # config location
@@ -78,6 +79,12 @@ if [ -z ${conda_exists+x} ]; then
     . $prefix/etc/profile.d/conda.sh &&
     conda activate &&
     conda_exists=1
+  elif [ -f "$miniforge3/bin/conda" ]; then
+    # look for a miniforge3 release
+    . $miniforge3/etc/profile.d/conda.sh &&
+    prefix=$miniforge3
+    conda activate &&
+    conda_exists=1
   elif [ -f "$berryconda/bin/conda" ]; then
     # look for a berryconda release
     . $berryconda/etc/profile.d/conda.sh &&
diff --git a/bin/unix-update.sh b/bin/unix-update.sh
index 0c902d1..9daddfe 100644
--- a/bin/unix-update.sh
+++ b/bin/unix-update.sh
@@ -12,6 +12,7 @@ if [[ "$arch" == "armv"* ]]; then release='berryconda3'; else release='miniconda
 # conda install location:
 prefix="$HOME/$release"         # $HOME/miniconda3 is default location
 full="$HOME/anaconda3"          # full release install location
+miniforge3="$HOME/miniforge3"   # MiniForge3
 berryconda="$HOME/berryconda3"  # berryconda install location
 miniconda="$HOME/miniconda3"    # miniconda install location
 config="$HOME/.config/rsudp"    # config location
@@ -60,6 +61,12 @@ if [ -z ${conda_exists+x} ]; then
     . $prefix/etc/profile.d/conda.sh &&
     conda activate &&
     conda_exists=1
+  elif [ -f "$miniforge3/bin/conda" ]; then
+    # look for a miniforge3 release
+    . $miniforge3/etc/profile.d/conda.sh &&
+    prefix=$miniforge3
+    conda activate &&
+    conda_exists=1
   elif [ -f "$berryconda/bin/conda" ]; then
     # look for a berryconda release
     . $berryconda/etc/profile.d/conda.sh &&
diff --git a/unix-install-rsudp.sh b/unix-install-rsudp.sh
index 2650fec..337bfef 100644
--- a/unix-install-rsudp.sh
+++ b/unix-install-rsudp.sh
@@ -7,6 +7,7 @@ if [[ "$arch" == "armv"* ]]; then release='berryconda3'; else release='miniconda
 # conda install location:
 prefix="$HOME/$release"         # $HOME/miniconda3 is default location
 full="$HOME/anaconda3"          # full release install location
+miniforge3="$HOME/miniforge3"   # MiniForge3
 berryconda="$HOME/berryconda3"  # berryconda install location
 miniconda="$HOME/miniconda3"    # miniconda install location
 config="$HOME/.config/rsudp"    # config location
@@ -27,6 +28,12 @@ if [ -z ${conda_exists+x} ]; then
     . $prefix/etc/profile.d/conda.sh &&
     conda activate &&
     conda_exists=1
+  elif [ -f "$miniforge3/bin/conda" ]; then
+    # look for a miniforge3 release
+    . $miniforge3/etc/profile.d/conda.sh &&
+    prefix=$miniforge3
+    conda activate &&
+    conda_exists=1
   elif [ -f "$berryconda/bin/conda" ]; then
     # look for a berryconda release
     . $berryconda/etc/profile.d/conda.sh &&
diff --git a/unix-start-rsudp.sh b/unix-start-rsudp.sh
index 6004263..87d91b7 100644
--- a/unix-start-rsudp.sh
+++ b/unix-start-rsudp.sh
@@ -10,11 +10,14 @@ if [[ "$arch" == "armv"* ]]; then release='berryconda3'; else release='miniconda
 # conda install location:
 prefix="$HOME/$release"         # $HOME/miniconda3 is default location
 full="$HOME/anaconda3"          # full release install location
+miniforge3="$HOME/miniforge3"   # MiniForge3
 
 if [ -f "$prefix/etc/profile.d/conda.sh" ]; then
   . $prefix/etc/profile.d/conda.sh
 elif [ -f "$full/etc/profile.d/conda.sh" ]; then
   . $full/etc/profile.d/conda.sh
+  elif [ -f "$miniforge3/bin/conda" ]; then
+  . $miniforge3/etc/profile.d/conda.sh
 else
   echo "Could not find an anaconda installation. Have you used the installer script to install rsudp, or is anaconda in a nonstandard location?"
   exit 2
@@ -25,4 +28,4 @@ mkdir -p /tmp/rsudp
 touch /tmp/rsudp/rsudp.log
 pip install $( cd "$( dirname "${BASH_SOURCE[0]}" )" >/dev/null 2>&1 && pwd ) &>/tmp/rsudp/rsudp.log
 echo "Done."
-rs-client -s $settings 2> >(tee -a /tmp/rsudp/rsudp.log >&2 2>&1)
\ No newline at end of file
+rs-client -s $settings 2> >(tee -a /tmp/rsudp/rsudp.log >&2 2>&1)

chaunceygardiner, 2024-08-03:
Interesting, my comment about getting this all working with the latest Raspberry OS 64-bit applies to the develop branch!  And the diff is against the develop branch.  Everything is running fine even though that branch hasn’t changed since last year.

chaunceygardiner, 2024-08-03:
OK, I’ve now tried the following branch:
update/python3.12

Everything I wrote above, applies to this branch.  The install doesn’t work because it attempts to install 32-bit miniconda.  If I install Miniforge3, and change the scripts to look for it, all works beautifully.

chaunceygardiner, 2024-08-03:
I did the work to have the install detect aarch64 and install miniforge3 in that case.  I’ve submitted the following pull request to your update/python3.12 branch:

https://github.com/raspishake/rsudp/pull/67

I’ve tested it on the latest 64-bit Raspberry OS.  I hope it helps, but I’m OK if you don’t merge it.

dmitryint, 2024-08-29:
Thanks for that, I'll add it. I plan to adapt and test the 32 bit Pi as well and try to do something with macOS. And then make a new release.

dmitryint, 2024-09-06:
[Eurasian](https://github.com/raspishake/rsudp/releases/tag/2.0.0)

closed 2025-01-16 by dmitryint