raspishake/rsudp #74: Installation issues

KampersandR, 2024-11-20:
Pi 5, up to date, 64 bits.

I’ve downloaded the unix-install shell scripts, put the rsudp one in ~ and the unix-install.sh in ~/bin.

Invoking either script ultimately results in the error:
Installing rsudp…
ERROR: Directory ‘/home/pi’ is not installable. Neither ‘setup.py’ nor ‘pyproject.toml’ found.

I’ve tried several suggestions from over the years via google, including running the scripts from a venv, and end up at the same error.

Please advise if ant logs or other info is wanted. 

Thanks for any pointers and/or fixes!

navneetercse, 2024-11-20:
https://forums.raspberrypi.com/viewtopic.php?t=51958

check this thread, there is some issue with your home directory

dmitryint, 2025-01-14:
Hello. There have been no new messages here for a long time. Do you have any more questions about this issue? If so, reopen this issue.

KampersandR, 2025-01-21:
Just a folowup... Nothing was wrong with the home directory. I chased down and removed every instance and fragment of Conda and rsudp and reinstalled with a fresh clone from github. Had to do a "conda install pyqt", and the plugin wayland can't be found, but that's another issue. Seems to be running on the Pi 5 now :-)