raspishake/rsudp #82: RuntimeWarning opn Bot.send_message in c_telegram.py

Lallirafn, 2025-03-07:
When trying to send alerts to telegram, this error is thrown:

`2025-03-07 23:00:04 [PlotAlert] Event time: 2025-03-07 23:00:04.18 UTC
C:\ProgramData\miniconda3\envs\rsudp2\Lib\site-packages\rsudp\c_telegram.py:165: RuntimeWarning: coroutine 'Bot.send_message' was never awaited
  self._when_alarm(d)
RuntimeWarning: Enable tracemalloc to get the object allocation traceback`

This was fixed last time with temporarily pinning the telegram bot version. I had a fix in issue #53 and even sent a pull request with the fix, but it was never implemented or looked at I think. Maybe its time to revisit and implement a permanent fix?

gpetricca, 2025-03-13:
This has been recreated.

In progress.

rankinstudio, 2025-03-29:
Can confirm the new asyncio code for telegram messaging in #54 works, but still no pngs saving or sending.

rankinstudio, 2025-04-03:
#85 fixes plotting and telegram issues. Forked here: https://github.com/rankinstudio/rsudp-fixes

gpetricca, 2025-04-23:
Fixed in the latest release: [Iberian](https://github.com/raspishake/rsudp/releases/tag/2.2.1).

closed 2025-04-23 by gpetricca

rankinstudio, 2025-04-23:
This doesn't look to have been implemented quite right. The self.timeout setting isn't being passed on.

The asyncio setup here is much more stable and implements upload timeouts:
https://github.com/rankinstudio/rsudp-fixes/blob/master/rsudp/c_telegram.py

You need to add both an upload and a write timeout or you'll get telegram timeout exceptions before the asyncio timeout value supplied:
```
                            response = await self.telegram.sendPhoto(chat_id=self.chat_id, photo=image,
                                                                     write_timeout=self.upload_timeout,
                                                                     read_timeout=self.upload_timeout)
```

` asyncio.run(asyncio.wait_for(self._when_img(d), self.upload_timeout))`

I did a lot of stability testing on that telegram setup, it's rock solid on a slow internet connection.

Cheers
David