raspishake/rsudp #83: plot routine does not save png of triggered events

Lallirafn, 2025-03-08:
Plot routine does not save png of triggered events anymore.

it is configured, and the plot saving routine is scheduled it seems, but it never triggers. 

`2025-03-08 09:53:48 [Plot] Event time: 2025-03-08 09:53:48.01 UTC
2025-03-08 09:53:48 [Plot] Saving png in about 42 seconds`

gpetricca, 2025-03-13:
This has been recreated.

In progress.

rankinstudio, 2025-03-24:
Also having this issue. No png images saving, telegram message fails - latest release:
2025-03-24 04:46:56 [Telegram id -79......] Telegram message: (Raspberry Shake station AM.RC1C0 - region: Eastern Arizona) Event detected at 2025-03-24 04:46:57.17 UTC Earthquake Detected - live feed  https://stationview.raspberryshake.org/#?net=AM&sta=RC1C0
C:\Users\david\anaconda3\envs\rsudp\Lib\site-packages\rsudp\c_telegram.py:165: RuntimeWarning: coroutine 'Bot.send_message' was never awaited
  self._when_alarm(d)

rankinstudio, 2025-03-29:
It appears this is because the save timer in c_plots.py.getq() is never getting higher than -1.

print("SAVE INFO", self.save_timer, self.save[0][0]) shows -1 for the save_timer for the entire trigger duration. Thus in main - if (self.save_timer > self.save[0][0]) is never true so self._eventsave() is never triggered.

rankinstudio, 2025-04-02:
Okay, doesn't seem to be much movement here. Got a hack working:

in c_plots.py.getq() 

`        if 'TERM' in str(d):
            plt.close()
            if 'SELF' in str(d):
                printM('Plot has been closed, plot thread will exit.', self.sender)
            self.alive = False
            self.controller.alive = False
            rs.producer = False

to:

`        if 'TERM' in str(d):
            plt.close()
            if 'SELF' in str(d):
                printM('Plot has been closed, plot thread will exit.', self.sender)
            self.alive = False
            self.controller.alive = False
            rs.producer = False

        if self.save:
            self.save_timer += 1`

Nothing is moving the self.save_timer forward. 

The telegram fix here works: https://github.com/raspishake/rsudp/pull/54/commits/12a3ec0da6b181c50d3b730e721e836ea6af8faa

With a few minor changes:
Change both `response = await self.telegram.sendPhoto(chat_id=self.chat_id, photo=image)`
to: `response = await asyncio.wait_for(self.telegram.sendPhoto(chat_id=self.chat_id, photo=image), 15)`

Change `asyncio.run(asyncio.wait_for(self._when_img(d), 3))`

to: `asyncio.run(self._when_img(d))`

I haven't tested these enough to know how resilient they are, seems to be working as before.

rankinstudio, 2025-04-03:
#85 fixes plotting and telegram issues. Forked here: https://github.com/rankinstudio/rsudp-fixes

gpetricca, 2025-04-23:
Fixed in the latest release: [Iberian](https://github.com/raspishake/rsudp/releases/tag/2.2.1).

closed 2025-04-23 by gpetricca