raspishake/rsudp #84: max sta/lta reached in alarm state is always 0

Lallirafn, 2025-03-08:
when an alarm is triggered, the max sta/lta reached is always 0. 

this used to show the maximum level reached inside the triggered event....

gpetricca, 2025-03-12:
Hello @Lallirafn, could you provide as many details of your RSUDP installation as possible so we can try to recreate this on our side?

I have tested the function with different STA/LTA thresholds, but I always get a non-zero value for the max STA/LTA. Thank you!

Lallirafn, 2025-03-13:
I am on windows 11 with miniconda environment, pulling the "master" branch for RSUDP. my alert settings are as follows:

`"alert": {
    "enabled": true,
    "channel": "HZ",
    "sta": 6,
    "lta": 30,
    "duration" : 0.9,
    "on_plot" : "on_main",
    "threshold": 0.9,
    "reset": 0.3,
    "highpass": 0.5,
    "lowpass": 15,
    "deconvolve": false,
    "units": "VEL"},`

gpetricca, 2025-03-13:
Thank you, I was able to recreate that.

In progress.

gpetricca, 2025-04-23:
Fixed in the latest release: [Iberian](https://github.com/raspishake/rsudp/releases/tag/2.2.1).

Thank you for your feedback.l

closed 2025-04-23 by gpetricca