raspishake/rsudp #86: Not Saving Screenshots

AdamsLab01, 2025-04-20:
Hello, 

I've got rsudp installed on a Pi 4. Everything looks to be working correctly, I'm getting a waveform and spectrogram and when I set it to write I get files save in the output_dir. However, when an alert it triggered, I'm not getting a screenshot saved. 

My  rsudp_settings.json:

`
{
  "settings": {
    "port": 8888,
    "station": "S8B94",
    "output_dir": "/home/pi/rsudp_data/",
    "debug": true
  },
  "printdata": {
    "enabled": false
  },
  "write": {
    "enabled": false,
    "channels": [
      "all"
    ]
  },
  "plot": {
    "enabled": true,
    "duration": 90,
    "refresh_interval": 0,
    "spectrogram": true,
    "fullscreen": true,
    "kiosk": true,
    "eq_screenshots": true,
    "channels": [
      "all"
    ],
    "filter_waveform": false,
    "filter_spectrogram": false,
    "filter_highpass": 0.7,
    "filter_lowpass": 2.0,
    "filter_corners": 4,
    "spectrogram_freq_range": false,
    "upper_limit": 15.0,
    "lower_limit": 0.0,
    "logarithmic_y_axis": false,
    "deconvolve": true,
    "units": "CHAN"
  },
  "forward": {
    "enabled": false,
    "address": [
      "192.168.1.254"
    ],
    "port": [
      8888
    ],
    "channels": [
      "all"
    ],
    "fwd_data": true,
    "fwd_alarms": false
  },
  "alert": {
    "enabled": true,
    "channel": "HZ",
    "sta": 6,
    "lta": 30,
    "duration": 0.0,
    "threshold": 1,
    "reset": 0.2,
    "highpass": 0.8,
    "lowpass": 9,
[... 167 lines cut ...]
2025-04-20 06:34:05 TESTING IMGPATH message             : FAIL
2025-04-20 06:34:05 TESTING TERM message                : PASS
2025-04-20 06:34:05 TESTING pydub dependencies          : PASS
2025-04-20 06:34:05 TESTING matplotlib backend          : PASS
2025-04-20 06:34:05 TESTING plot                        : FAIL
2025-04-20 06:34:05 TESTING miniSEED write              : PASS
2025-04-20 06:34:05 TESTING miniSEED data               : PASS
2025-04-20 06:34:05 TESTING print data                  : PASS
2025-04-20 06:34:05 TESTING alert trigger on            : PASS
2025-04-20 06:34:05 TESTING alert trigger off           : PASS
2025-04-20 06:34:05 TESTING play sound                  : PASS
2025-04-20 06:34:05 TESTING screenshot exists           : FAIL
2025-04-20 06:34:05 TESTING Twitter text message        : PASS
2025-04-20 06:34:05 TESTING Twitter image message       : FAIL
2025-04-20 06:34:05 TESTING Telegram text message       : PASS
2025-04-20 06:34:05 TESTING Telegram image              : FAIL
2025-04-20 06:34:05 TESTING forwarding                  : PASS
2025-04-20 06:34:05 TESTING RSAM transmission           : PASS
2025-04-20 06:34:05 TESTING [Main] Shutdown successful.
`
Not sure why plot failed, as it displayed ok, IMGPATH message fail might indicate something but I'm not sure what.

Any help is appreciated. 

Adam

rankinstudio, 2025-04-22:
I haven't heard from any of the current maintainers for a while. I fixed that issue and a few others in this recent fork: https://github.com/rankinstudio/rsudp-fixes

gpetricca, 2025-04-23:
Fixed in the latest release: [Iberian](https://github.com/raspishake/rsudp/releases/tag/2.2.1).

Thank you for the detailed feedback and participation!

closed 2025-04-23 by gpetricca